thanks4opensource/buck50 #1: Black pill

beriberikix, 2020-11-14:
Have you considered adding support for the so called "[black pill](https://www.cnx-software.com/2019/12/24/stm32-black-pill-board-features-stm32f4-cortex-m4-mcu-optional-spi-flash/)"? The two STM32F4 variants are much more capable than the F1, and still pretty inexpensive @ $3. If naming is a problem you could call it, "buck50x2" 😄

thanks4opensource, 2020-11-14:
It's possible, but no immediate plans. I actually do have a couple of "black pills" (in some variant or another) but haven't gotten around to playing with them yet.

As you point out, the price/performance ratio is tempting, as opposed to other suggestions I've had suggesting ports to much more expensive (although also much more capable) dev boards. The clock speed increase from the "Blue Pill"'s 72MHz to 84 or 100 isn't that great, but combined with the F4 architecture and the ART accelerator the performance delta is probably greater than the raw numbers suggest. The biggest win would be 64 or 128K of RAM, but as "buck50" uses sample memory very efficiently this would only affect long captures with continuously changing signals.

The biggest problem is USB. Most of ST's newer medium-and-above level chips use their new USB OTG peripheral which is completely different than the one in the F103. I've been very vocal in my criticisms of ST's USB libraries -- my  https://github.com/thanks4opensource/papoon_usb replacement took a very long time to write (deciphering ST's incomplete documentation and horribly obfuscated example code). The  hope was that if and when I do try to port to the new version, the painfully gained experience and the fact that more functions have been moved into hardware would make the process easier the second time around, but I've been warned by knowledgeable people that this won't be the case ("it's even worse than the 103" was the statement).

Other than that major stumbling point, a port should (famous last words) be fairly easy. Just lots of small details, and even more testing. The other peripherals, including I2C (which was also a development nightmare, getting full slave-xmit-to-master with repeated START condition working) seem to be identical or only slightly changed. How about "3-bucks-even" for the name? ;)

beriberikix, 2020-11-14:
That's definitely a better name 😄 

Ya, I've heard more than one horror story about ST's USB. Have you seen [TinyUSB](https://github.com/hathach/tinyusb)? I'd imagine it would help with the port.

thanks4opensource, 2020-11-14:
Thanks for the TinyUSB link. Ironically, it was one of the many libraries/sources/examples I looked at when I started [papoon_usb](https://github.com/thanks4opensource/papoon_usb) 18 months ago, At the time it either didn't support the F103, or it used ST's obfuscated HAL as a back end, or I maybe I just couldn't figure out what it was doing with that series of ST's chips. ;)

I just now did a `git pull` on my local copy and it's changed radically. The new [dcd_stm32_fsdev.c]( https://github.com/hathach/tinyusb/blob/master/src/portable/st/stm32_fsdev/dcd_stm32_fsdev.c) file has F103 support implemented using bare-metal, register level code. That would have been a huge help back when I did "papoon" -- as its README.md details, none of the bare-metal examples I found worked, and I made an unfortunate choice by starting with possibly the worst ST pre-HAL code for my own development.

TinyUSB did support the OTG peripheral back then, but it was of no use to me at the time so I put the whole library aside. A quick browse now shows the code to be very clean, so maybe (wonder of wonders!) I could use it as starting point for a "papoon2", write a wrapper around it, or (horrors!) us it as-is. To my (imagined) detractors: See? I sometimes do use code that I didn't write myself to my own peculiar standards. ([libusb](https://github.com/libusb/libusb) on the host side is a beautifully written piece of software engineering with an ideal, well thought out API that's a joy to use.)

Thanks  again for the pointer. Maybe there's hope yet. ;)

beriberikix, 2020-11-15:
BTW, I posted a link to the repo [on Reddit](https://www.reddit.com/r/embedded/comments/ju29id/buck50_blue_pill_stm32f103_logic_analyzer_and_more/) and people seem to like it :)

thanks4opensource, 2020-11-15:
Thanks @beriberikix. If people use the code it makes the effort creating it worthwhile.

I joined Reddit and answered a few of the questions there.

Closing this issue. Anything further, don't hesitate to open a new one. Especially to report bugs -- I know there must be some!

closed 2020-11-15 by thanks4opensource

fanoush, 2020-12-13:
I almost created identical issue about suggesting WeAct Black Pill board but then found this closed issue :-)

BTW their github is here https://github.com/WeActTC/MiniF4-STM32F4x1 and recently they upgraded cheaper F401 board to 96KB/512KB, current price $2.89 + shipping

so just a +1 post to subscribe to this topic and thank you for buck50 :-)

thanks4opensource, 2020-12-13:
Thanks, @fanoush. I didn't know about the WeAct github. I wish I had because unfortunately I've bought  boards in the past from some of the vendors they list as being pirates. :(

fanoush, 2020-12-13:
well, they publish the board schematics so others are making more or less identical clones, calling them pirates is a bit exaggerated IMO, but buying directly from them may be good to support them and get latest revision, they also sell version with 8Mhz crystal (if someone prefers it to the 25Mhz) and another with presoldered SPI flash. And btw they also designed a bit more expensive blue pill https://github.com/WeActTC/BluePill-Plus with real STM chip (and usb-c and hopefully fixed resistor) but not sure if it is worth it for the price when compared to F401.