thanks4opensource/buck50 #9: Building with gcc-arm-none-eabi-10-2020-q4-major fails

kapacuk, 2021-03-10:
Here's the error message:

```
arm-none-eabi-ld -o buck50.elf -static -M --print-memory-usage			\
	    buck50.o usb_dev.o usb_dev_cdc_acm.o buck50_asm.o stm32f103_flash_init.o    			\
	    -T src/low_stack_flash.ld		\
     > buck50.elf.map
arm-none-eabi-ld: stm32f103_flash_init.o: in function `init':
/home/dima/repos/buck50/build/init/stm32f103_flash_init.c:55: undefined reference to `memcpy'
arm-none-eabi-ld: /home/dima/repos/buck50/build/init/stm32f103_flash_init.c:61: undefined reference to `memset'
arm-none-eabi-ld: buck50.o: in function `live()':
/home/dima/repos/buck50/build/src/buck50.cxx:3216: undefined reference to `memset'
arm-none-eabi-ld: /home/dima/repos/buck50/build/src/buck50.cxx:3203: undefined reference to `memset'
arm-none-eabi-ld: /home/dima/repos/buck50/build/src/buck50.cxx:3190: undefined reference to `memset'
make: *** [Makefile:49: buck50.elf] Error 1

```
When I added "-fno-builtin" to CC_ARM_FLAGS it worked fine.

thanks4opensource, 2021-03-11:
Yes, somewhere between gcc-arm-none-eabi-9-2020-q2-update and gcc-arm-none-eabi-10-2020-q4-major the compiler got more aggressive at detecting and replacing memset-like code with calls to its own optimized versions.

I'll push up a change with the extra Makefile flag plus a small fix for a new warning after I have a chance to test it. Until then, doing as you suggest (and ignoring the warning) will work. Thanks for reporting this issue.

fmkit, 2021-03-23:
compiled with Platformio (default 2017 GCC): 
arm-none-eabi-g++ -c  -std=c++17 -Wall -Wextra -g3 -mthumb -mcpu=cortex-m3 -fno-exceptions -fno-unwind-tables -I. -I./init -I./include/arm -I./include/st/32f -I./include/thanks4opensource -I./util/stm32f10_12357xx -O2 -DUSB_DEV_FLASH_WAIT_STATES=2 -DINLINE_DECL=inline -DINLINE_ATTR='__attribute__((always_inline))' -fno-threadsafe-statics usb_dev.cxx -o usb_dev.o
arm-none-eabi-g++ -c  -std=c++17 -Wall -Wextra -g3 -mthumb -mcpu=cortex-m3 -fno-exceptions -fno-unwind-tables -I. -I./init -I./include/arm -I./include/st/32f -I./include/thanks4opensource -I./util/stm32f10_12357xx -O2 -DUSB_DEV_FLASH_WAIT_STATES=2 -DINLINE_DECL=inline -DINLINE_ATTR='__attribute__((always_inline))' -fno-threadsafe-statics usb_dev_cdc_acm.cxx -o usb_dev_cdc_acm.o
arm-none-eabi-as -g3 -mthumb -mcpu=cortex-m3 -march=armv7-m  buck50_asm.s -o buck50_asm.o
arm-none-eabi-gcc -c  -Wall -Wextra -g3 -mthumb -mcpu=cortex-m3 -fno-exceptions -fno-unwind-tables -I. -I./init -I./include/arm -I./include/st/32f -I./include/thanks4opensource -I./util/stm32f10_12357xx -DWEAK_FUNCTIONS -O2 stm32f103_flash_init.c -o stm32f103_flash_init.o
arm-none-eabi-ld -o buck50.elf -static -M --print-memory-usage			\
	    buck50.o usb_dev.o usb_dev_cdc_acm.o buck50_asm.o stm32f103_flash_init.o    			\
	    -T low_stack_flash.ld		\
     > buck50.elf.map
Memory region         Used Size  Region Size  %age Used
           FLASH:       21452 B        64 KB     32.73%
             RAM:        1176 B        20 KB      5.74%
arm-none-eabi-objcopy -O ihex buck50.elf buck50.hex
arm-none-eabi-objcopy -O binary buck50.elf buck50.bin