thanks4opensource/buck50 #12: Can't configure trigger for edge triggering

nexysm, 2022-08-14:
I am using `logic` command in order to record a digital signal of about 19.2 kbaud and I want to set trigger in falling edge.
I tried this:
`$1.50: trigger
trigger 0=0xxxxxxx-1-0 st1
trigger 1=1xxxxxxx-0-1 st2`

`logic` doesn't capture anything.
I also tried to reverse the logic:
`$1.50: trigger
trigger 0=1xxxxxxx-1-0 st1
trigger 1=0xxxxxxx-0-1 st2`

Also, no 0 catches samples.

`$1.50: trigger
trigger 0=0xxxxxxx-0-1 st1
trigger 1=1xxxxxxx-1-0 st2`

Captures the signal, but it complains about: `Infinite triggers "pass" loop: 1 -> 1`. In this case the digital signal is broken and not correctly shown as with the next command.

If I try with this trigger:
`trigger 0=0xxxxxxx-0-0 st1`
I get samples and I see the signal.

However, `trigger 0=1xxxxxxx-0-0 st1` doesn't trigger logic although I would expect the same behavior as before.

thanks4opensource, 2022-08-18:
Trigger commands must be written one per line. Your second example, if entered correctly, should trigger on a falling edge on input PB11.

    $1.50: trigger 0=1xxxxxxx-1-0 st1
    $1.50: trigger 1=0xxxxxxx-0-1 st2

Enter a plain `trigger` command to list the current trigger states and confirm that they're what you want.

nexysm, 2022-08-19:
For some reason, the text is shown in a single line, but I copied the output from a plain trigger.
THe first and the second cases doesn't capture anything. Only the third one.