thanks4opensource/buck50 #13: Trouble with Raspberry PI

arpruss, 2022-10-01:
I uploaded buck50.bin using the STM demonstrator uploader on Windows (which I normally use for uploading bootloaders to blue pills). I then plugged the pill into my RPI 3B+, and I got this from dmesg:
`
[   44.366685] usb 1-1.1.2: new full-speed USB device number 9 using dwc_otg
[   44.466662] usb 1-1.1.2: device descriptor read/64, error -32
[   44.686658] usb 1-1.1.2: device descriptor read/64, error -32
[   44.906681] usb 1-1.1.2: new full-speed USB device number 10 using dwc_otg
[   45.006680] usb 1-1.1.2: device descriptor read/64, error -32
[   45.226743] usb 1-1.1.2: device descriptor read/64, error -32
[   45.346919] usb 1-1.1-port2: attempt power cycle
[   46.006713] usb 1-1.1.2: new full-speed USB device number 11 using dwc_otg
[   46.446703] usb 1-1.1.2: device not accepting address 11, error -32
`

arpruss, 2022-10-01:
Ah, I think it's a clone issue. I tried it with an older blue pill and it worked fine.

thanks4opensource, 2022-10-03:
That's good to hear.

If you're curious and have the time, you might want to try some other USB firmware (like an ST HAL demo) that's known to work with genuine hardware and see what it does on the clone. Buck50 is based on my [papoon_usb](https://github.com/thanks4opensource/papoon_usb)  library, and although that's worked for me and many others (note how few issues have been logged against it or Buck50), it's a complete clean-sheet design and I can imagine edge cases exist on the clones that ST's implementation works around.

That said, I think it's more likely that the clone's USB harware doesn't sufficiently match ST's at the register level and all libraries will fail on it in at least some scenarios.

arpruss, 2022-10-03:
Some things work on the clone, and others don't. 

I normally use Arduino with the libmaple-derived core with blue pills. When I tried to activate the RTC in internal oscillator mode, it just hung (though it worked fine on another blue pill in my stash). On the other hand, some other things worked fine. For instance, libmaple's USB serial transmits fine (I didn't check receiving).